fsantini/python-e3dc #1: Not an issue...

VDRainer, 2018-11-08:
...just want to say THANKS for this API.
Works great for me.

fsantini, 2018-11-09:
Wow! Thank you for taking your time to leave a comment! It's great to know that this works for someone else! I've been using it for the last year at my home and never found any major problem with it. However, as you can imagine, the implementation required some amount of guesswork so there might be problems in some edge cases. Let me know if you have any issue or if you would like some other feature added!

closed 2018-11-09 by fsantini

VDRainer, 2018-11-09:
> the implementation required some amount of guesswork

Yes, as i'm not a python dev it took some trial and error to get it to work. :blush:
At least i edited https://github.com/fsantini/python-e3dc/blob/9794cfe452438496fbef98ede2cbac2426607e97/e3dc/_e3dc.py#L206-L233 to my needs.
```
        sys_time = self.rscp.sendRequest( ('INFO_REQ_TIME', 'None', None) )[2]
        power_pv = self.rscp.sendRequest( ('EMS_REQ_POWER_PV', 'None', None) )[2]
        power_add = self.rscp.sendRequest( ('EMS_REQ_POWER_ADD', 'None', None) )[2]
        power_home = self.rscp.sendRequest( ('EMS_REQ_POWER_HOME', 'None', None) )[2]
        power_grid = self.rscp.sendRequest( ('EMS_REQ_POWER_GRID', 'None', None) )[2]
        power_bat = self.rscp.sendRequest( ('EMS_REQ_POWER_BAT', 'None', None) )[2]
        bat_soc = self.rscp.sendRequest( ('EMS_REQ_BAT_SOC', 'None', None) )[2]
        self_consumption = self.rscp.sendRequest( ('EMS_REQ_SELF_CONSUMPTION', 'None', None) )[2]
        autarky = self.rscp.sendRequest( ('EMS_REQ_AUTARKY', 'None', None) )[2]
        emergency_power_status = self.rscp.sendRequest( ('EMS_REQ_EMERGENCY_POWER_STATUS', 'None', None) )[2]
        sw_release = self.rscp.sendRequest( ('INFO_REQ_SW_RELEASE', 'None', None) )[2]
        serial_number = self.rscp.sendRequest( ('INFO_REQ_SERIAL_NUMBER', 'None', None) )[2]

        #wb = self.rscp.sendRequest( ('EMS_REQ_POWER_WB_ALL', 'None', None) )[2]

        #home = solar + grid - bat - wb # make balance = 0

        if not keepAlive:
            self.rscp.disconnect()

        outObj = {
            'sys_time': datetime.datetime.fromtimestamp(sys_time).isoformat(),
            'power_pv': power_pv,
            'power_add': power_add,
            'power_home': power_home,
            'power_grid': power_grid,
            'power_bat': power_bat,
            'bat_soc': bat_soc,
            'self_consumption': int(round(self_consumption)),
            'autarky': int(round(autarky)),
            'emergency_power_status': emergency_power_status,
            'sw_release': sw_release,
            'serial_number': serial_number
            }
```
Then i modified your example script to publish the values to mqtt.
```
import json
from e3dc import E3DC
import paho.mqtt.client as mqtt
from time import sleep

# E3DC
TCP_IP = '192.168.0.58'
USERNAME = 'Username'
PASS = 'password'
KEY = 'key'

# MQTT
BROKER = '127.0.0.1'
TOPIC = 'home/e3dc/rscp'

looptime = 30

client = mqtt.Client("e3dc_poll")
client.connect(BROKER)

e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress = TCP_IP, key = KEY)

while True:
    try:
        message = e3dc.poll()
        print message
        client.publish(TOPIC, payload=json.dumps(message), qos=0, retain=False)
        sleep(looptime)
    except:
        print 'bad request, trying again'
```
The 'try/except' was needed because sometimes there were communication error, or problems with unpacking the response when strings ('INFO_REQ_SW_RELEASE','INFO_REQ_SERIAL_NUMBER')are returned.