fsantini/python-e3dc #2: Web Connection Error

hismastersvoice, 2019-06-15:
Hi,

If I use the local connection everthing works fine for me. 
But if I try to use the web connection I get the follow error. I can´t find the failure :(

web connection
Traceback (most recent call last):
  File "test.py", line 12, in <module>
    print e3dc.poll()
  File "/python-e3dc/e3dc/_e3dc.py", line 181, in poll_ajax
    raw = self.poll_ajax_raw()
  File "/python-e3dc/e3dc/_e3dc.py", line 152, in poll_ajax_raw
    raise PollError("Error polling: %d" % (jsonResponse['ERRNO']))
e3dc._e3dc.PollError: Error polling: 6632

Can you help me.
I need the connection for turn the switches.
Thanks...

fsantini, 2019-06-15:
They must have changed something in the web protocol... I don't use it so I didn't notice. I'll try to investigate the issue!

hismastersvoice, 2019-06-20:
Thanks for investigate....
The idea behind your software is what I´m serching for a long time

fsantini, 2019-06-21:
Recently the portal required a window-id to be defined for ajax requests. I added it in the code and it should work now.

closed 2019-06-21 by fsantini

hismastersvoice, 2019-06-23:
THANKS !!!
Works again...