fsantini/python-e3dc #4: Download csv file from Website

linderol, 2019-11-25:
Hi, great library. Thanks for building it. I only have to get it run as I'm not a developer. 

My question is: Is there a way to request the csv file automatically? On the website you can choose the timeframe, the kpi and hit the export button. I'd like to automate this process. 

As I understand your lib, it calls always the current status, right? so if the call doesn't work for however reason, I will loose data. Please correct me if I'm wrong. I see the benefit of using the CSV that I already have aggregated data of 15min.

Thanks for answering. 

Best, 
Oliver

fsantini, 2019-11-25:
Hi thanks for your comment. The idea behind this library is that you have real-time control, that is, you monitor how much power you have available and you control smart appliances. I am not sure what your use-case would be: logging your production/consumption?

linderol, 2019-11-25:
Thanks for coming back so quickly. My use case is more to analyse my behavior during the day, see trends to optimize things, etc. rather than having real time control. 

I quickly build some views (attached screnshot) based on the historical data downloaded in the csv. If I now could automate that process without having to login, choose the timeframe ann export the data all over again, would myke my life easier. :-)
[image]

fsantini, 2019-11-25:
I had a quick look, it's rather complicated as everything is generated on-the-fly in Javascript. At the moment I don't think I can tackle that, sorry!

seh-len, 2020-08-25:
It should also be possible to get these data out of the e3dc via rscp using the TAG_DB_REQ_HISTORY_* tags. I tried to get these data but ended with an error. Maybe someone is able to work this out.

closed 2022-01-21 by vchrisb