fsantini/python-e3dc #6: Python 3 Support

vchrisb, 2020-01-07:
I gave the python3 branch a test and it works partially:

I additionally needed to installe modules `websocket-client` and `python-dateutil`.
`e3dc.poll()` with web connection does work with using a binary string for the password.
`e3dc.poll_switches()` does get a timeout.

For local it does not work at all:
```
  File "/Users/cbanck/Documents/GitHub/python-e3dc/e3dc/_e3dc_rscp_local.py", line 28, in __init__
    self.password = password.encode('utf-8')
AttributeError: 'bytes' object has no attribute 'encode'
```

fsantini, 2020-01-07:
Thanks for the quick feedback! I tried it at home and it was running, I'll look into this to find out what I did differently.

vchrisb, 2020-01-07:
Update: `local` pull works with a non-binary password.
So maybe also adding a `.encode('utf-8')` to https://github.com/fsantini/python-e3dc/blob/923bf98b25416f6d63cf5b88421491ed091d3c03/e3dc/_e3dc_rscp.py#L92 does make it consistent?

vchrisb, 2020-01-09:
I've added a few more capabilities in #8 

I only struggle to read `PM_REQ_DATA`

I use 
```
self.sendRequest( ('PM_REQ_DATA', 'Container', [ ('PM_INDEX', 'Uint16', 0), ('PM_REQ_POWER_L1', 'None', None)]), keepAlive )
```
But I always get back:
```
('PM_DATA', 'Container', [('PM_INDEX', 'Uint16', 0), ('PM_DEVICE_STATE', 'Error', 'RSCP_ERR_NOT_AVAILABLE')])
```
though I'm even not requesting `PM_DEVICE_STATE`.
According to this https://github.com/Eba-M/E3DC-Control/blob/master/RscpExampleMain.cpp#L593 it should work like that.
Any idea what I might do wrong?

vchrisb, 2020-01-09:
I've implemented a REST API on top of the changes in #8 
https://github.com/vchrisb/e3dc-rest

So I'm only struggling with `PM_DATA` still.

fsantini, 2020-01-12:
> So I'm only struggling with `PM_DATA` still.

Hi! Great work! I'll also try your PR and merge it later. In the meantime, I have good and bad news for you:

`print( e3dc.sendRequest( ('PM_REQ_DATA', 'Container', [ ('PM_INDEX', 'Uint16', 0), ('PM_REQ_POWER_L1', 'None', None)])) )`

for me returns

`('PM_DATA', 'Container', [('PM_INDEX', 'Uint16', 0), ('PM_POWER_L1', 'Double64', 2897.0)])`

I'm testing it with the local RSCP connection, on the python3 branch as it is now (i.e. with your first commits merged)

vchrisb, 2020-01-12:
I guess you have a `S10`? I have a `S10 mini`, maybe I need to read it differently...

fsantini, 2020-01-13:
Yes that must be it... Have you tried a different PM_INDEX maybe?

vchrisb, 2020-01-13:
I tried multiple options but wasn't successful. I just opened a Ticket with E3DC, lets see if they can help.

vchrisb, 2020-01-22:
I added quite a bit of error handling and retries as the connection was failing after a few days of continuous running with my e3dc-rest implementation.
I also modified more functions to make it more consistent and hopefully easier to use.

fsantini, 2020-01-22:
Yes I've seen! That sounds very reasonable but it breaks some of my high-level code... But I agree that it's nicer your way.

Regarding error-checking, I know, I have the same problem, I usually catch it higher up, but your way is great.

Can you also change the thing I marked in the review? Maybe convert the bytes to string only if the format tag is 'CString'... I wouldn't want to break other things.

vchrisb, 2020-01-27:
For the S10 Mini the external power meter has a value of `6`. I didn't test that far...

vchrisb, 2020-03-26:
@fsantini I added power meter support in #13 
Can you please validate that it is working for a S10 also?

closed 2020-06-07 by vchrisb