fsantini/python-e3dc #12: Setting battery charge power

DomiJV, 2020-02-28:
Hi

First of all thank you for your amazing work. It made interfacing the e3dc so much easier!

Currently I'm trying to limit the battery charge power. But I can't get it to work. I use the local connection.
Quering the data works fine. So the connection shouldn't be the problem. 

This is the code I'm using:
` y = e3dc.sendRequest( ("SET_POWER_SETTINGS", "Container", [
        ("MAX_CHARGE_POWER", "Uint32", 100)]), keepAlive=True)`

It always returns CommunicationError and doesnt change the setting.
Do you know what i'm doing wrong?

fsantini, 2020-02-28:
Hi! Are you using the python3 branch? You can use the high-level interface in the e3dc class `set_power_limits`.
Otherwise you can manually send a request like this:
`res = self.sendRequest(("EMS_REQ_SET_POWER_SETTINGS", "Container", [ ("EMS_POWER_LIMITS_USED", "Bool", True), ("EMS_MAX_DISCHARGE_POWER", "Uint32", max_discharge), ("EMS_MAX_CHARGE_POWER", "Uint32", max_charge), ("EMS_DISCHARGE_START_POWER", "Uint32", discharge_start) ]), keepAlive = keepAlive)`

DomiJV, 2020-02-29:
Great!
I didnt know that there was a high level function for that. 
It works well now.
Thank you

closed 2020-02-29 by DomiJV