fsantini/python-e3dc #14: Connection outside of LAN

Doriangaensslen, 2020-03-30:
Hi fsantini

Awesome what you did! Many thanks!
I have serveral VLAN's at Home. If i try to connect from another VLAN i can see the 3-Way TCP handshake, but after that, it drops my Connection (RST). However, in the same local network it works without problems. I wonder, if this works for anyone from another network.

All the best
Dorian

fsantini, 2020-03-31:
Hi thanks! If I remember correctly, the manual states that the RSCP connection only works within the same VLAN. You might try the web connection, but I haven't tested it for a while, so I don't know if it still works.

Doriangaensslen, 2020-03-31:
I would like it to stay in my own network. I don't like outside network traffic if its not 100% needed ;).
Can you point me to the direction where you've read that? i cant find anything about that.

fsantini, 2020-03-31:
The "RSCP-Beispielapplikation" document states that

> Für die externe Steuerung des Hauskraftwerks oder die Beschaffung von Hauskraftwerksdaten, wird von der Software des Hauskraftwerks ein TCP/IP- Server gestartet, auf den alle Teilnehmer im **lokalen Ethernet Subnetz** Zugriff haben

I also tried to connect through VPN at some point at it didn't work. I don't think that there is anything we can do from our side. A possible solution for you would be to place a small raspberry pi in the same subnet and let it run a server that communicates with the e3dc. You can have a look at the REST API from vchrisb https://github.com/vchrisb/e3dc-rest

Doriangaensslen, 2020-03-31:
Thanks, i've searched for hours because of that.
The sad thing is, in my manual for my S10 Pro isn't anything written about that.
Il guess it's a good idea for E3DC to point that out.

closed 2020-03-31 by Doriangaensslen

Doriangaensslen, 2020-03-31:
Ive opened a ticket for that.