fsantini/python-e3dc #15: web connection working on the python3 branch?

juamiso, 2020-04-21:
Hi, I am doing this for a good friend. Thanks for the possibilty:
Using:
`from e3dc import E3DC

TCP_IP = '192.168.0.31'
USERNAME = '---'
PASS = '---'
KEY = '---'
SERIALNUMBER = '---'

print ("web connection")
e3dc = E3DC(E3DC.CONNECT_WEB, username=USERNAME, password=PASS, serialNumber = SERIALNU>
# connect to the portal and poll the status. This might raise an exception in case of f>
print (e3dc.poll())
`

I get this output ` python3 alex.py
web connection
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/e3dc-0.4.0-py3.7.egg/e3dc/_e3dc.py", line 340, in sendRequest
  File "/usr/local/lib/python3.7/dist-packages/e3dc-0.4.0-py3.7.egg/e3dc/_e3dc_rscp_web.py", line 275, in connect
e3dc._e3dc_rscp_web.RequestTimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "alex.py", line 10, in <module>
    e3dc = E3DC(E3DC.CONNECT_WEB, username=USERNAME, password=PASS, serialNumber = SERIALNUMBER, isPasswordMd5=False)
  File "/usr/local/lib/python3.7/dist-packages/e3dc-0.4.0-py3.7.egg/e3dc/_e3dc.py", line 106, in __init__
  File "/usr/local/lib/python3.7/dist-packages/e3dc-0.4.0-py3.7.egg/e3dc/_e3dc.py", line 537, in get_system_info_static
  File "/usr/local/lib/python3.7/dist-packages/e3dc-0.4.0-py3.7.egg/e3dc/_e3dc.py", line 348, in sendRequest
e3dc._e3dc.SendError: Max retries reached
`

I've checked password and username like 10x and are ok. What am I doing wrong?

fsantini, 2020-04-21:
Hi! Sorry the web connection was never properly tested and it was indeed broken since the upgrade to python3. I think it should be fixed now, can you pull the latest version and try again?

juamiso, 2020-04-21:
Works  :+1: 
web connection
{'time': datetime.datetime(2020, 4, 21, 17, 32, 28, tzinfo=datetime.timezone.utc), 'sysStatus': '703', 'stateOfCharge': 100, 'production': {'solar': 609, 'grid': -165}, 'consumption': {'battery': 0, 'house': 444, 'wallbox': 0}}

closed 2020-04-22 by fsantini