fsantini/python-e3dc #16: please add possibility to read all dc strings

seh-len, 2020-04-27:
Hi,

please add the possibility to read all connected dc strings via get_pvi_data.

> req = self.sendRequest( ('PVI_REQ_DATA', 'Container', [
            ('PVI_INDEX', 'Uint16', 0),
            ('PVI_REQ_DC_VOLTAGE', 'UChar8', **0**),
            ('PVI_REQ_DC_CURRENT', 'UChar8', **0**),
            ('PVI_REQ_DC_POWER', 'UChar8', **0**),
            ('PVI_REQ_DC_VOLTAGE', 'UChar8', **1**),
            ('PVI_REQ_DC_CURRENT', 'UChar8', **1**),
            ('PVI_REQ_DC_POWER', 'UChar8', **1**),
            ]), keepAlive = False )

You can get the string count via PVI_REQ_DC_MAX_STRING_COUNT.

Thank you.
Kind regards,
Simon

fsantini, 2020-04-29:
Hi, sorry for the delay in replying. I don't understand what you are actually asking, as the code you sent does work. Would you like to see it implemented in a high-level function?

seh-len, 2020-04-29:
Hi, sorry, I forgot to mention that it should get integrated into get_pvi_data().

closed 2020-08-25 by seh-len