fsantini/python-e3dc #17: struct.error: unpack requires a buffer of 2 bytes

frefal, 2020-05-10:
Hi,
Trying to get this running against a S10 Pro. Not sure if this is a problem with my system, or the module. Any tip would be greatly appreciated !

Python 3.7.3

Installed it via:
- git clone
- pip3 install .

Using the sample provided from the readme,:

`from e3dc import E3DC
TCP_IP = 'x.x.x.x'
USERNAME = 'xx'
PASS = 'xx'
KEY = 'xx'
SERIALNUMBER = 'xx'

print("local connection")
e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress = TCP_IP, key = KEY)
print(e3dc.poll())
print(e3dc.poll_switches())`

**Executed:**
python3 test.py

**Result:**
local connection
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc_rscp_local.py", line 51, in sendRequest
    receive = self._receive()
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc_rscp_local.py", line 42, in _receive
    decData = rscpDecode( self.encdec.decrypt(data) )[0]
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_rscpLib.py", line 149, in rscpDecode
    magic = struct.unpack(magicCheckFmt, data[:struct.calcsize(magicCheckFmt)])[0]
struct.error: unpack requires a buffer of 2 bytes

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc.py", line 340, in sendRequest
    self.rscp.connect()
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc_rscp_local.py", line 79, in connect
    ('RSCP_AUTHENTICATION_PASSWORD', 'CString', self.password)]) )
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc_rscp_local.py", line 54, in sendRequest
    raise CommunicationError
e3dc._e3dc_rscp_local.CommunicationError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 18, in <module>
    e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress = TCP_IP, key = KEY)
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc.py", line 106, in __init__
    self.get_system_info_static(keepAlive=True)
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc.py", line 537, in get_system_info_static
    self.deratePercent  = round(self.sendRequest( ('EMS_REQ_DERATE_AT_PERCENT_VALUE', 'None', None), keepAlive = True  )[2] * 100)
  File "/usr/local/lib/python3.7/dist-packages/e3dc/_e3dc.py", line 348, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached

fsantini, 2020-05-12:
Mmm it seems that the system is returning some wrong data. I think that the only situation in which you could have this error is if the 'data' variable is smaller than 2 bytes. Maybe you can try troubleshooting by adding a print(len(data)) in rscpDecode before the statement that raised the error.

frefal, 2020-05-12:
it prints out 0 
any further idea would be greatly appreciated !

fsantini, 2020-05-12:
Maybe could you check the encryption key if it's correct?

frefal, 2020-05-12:
*facepalm* that is --- thanks!!

fsantini, 2020-05-12:
Lol. Glad that we found the culprit. I should try to catch the error and return a meaningful error message.

grothkopp, 2021-07-19:
Can someone help me where I find or set this encryption key? The only key I found is the cloud-registry key in "User Profile" but I get the same error as described here when I use that one.

vchrisb, 2021-07-19:
In German:
`Hauptmenü > Personalisieren > Benutzerprofil > RSCP Passwort`

grothkopp, 2021-07-19:
Thank you! Maybe someone could add that to the README.md to clarify "The encryption key as set under the preferences of the system"?

mdhom, 2022-03-02:
this info can be found in the readme (just checked) -> this issue can be closed I think?

mdhom, 2022-03-06:
#60 should tackle this issue

closed 2023-02-01 by vchrisb