fsantini/python-e3dc #18: Getting history data

steff333, 2020-05-24:
Im trying to read the history data with DB_REQ_HISTORY_DATA_DAY but can't get it to work.
The reason is to get the daily sums for production, grid etc.

Did you get it to work?
I checked for example  
e3dc.sendRequest(('DB_REQ_HISTORY_DATA_DAY', 'Container', [
        ('DB_REQ_HISTORY_TIME_START', 'Uint16', 0),
        ('DB_SUM_CONTAINER', 'None', None),]), keepAlive=True)
but always only get errors back.

Can you help?

fsantini, 2020-05-24:
I can try, but it seems that it's an error in the format of the request, and I have no clue about those. I only know how to handle the low-level communication. How did you come up with this request?

steff333, 2020-05-24:
Honestly, I tried to get access to the archive values - I'm missing a proper documentation from e3dc.
Is there a better way to retrieve e.g. the total amount of production by day?

It would be rather dull to read the data every second and calculate something manually.

steff333, 2020-06-01:
I think I fixed it and sent you a pull request for the python3 branch

vchrisb, 2020-06-07:
Thank you very much for your contribution. #20 
I already picked up your enhancements for [e3dc-rest](https://github.com/vchrisb/e3dc-rest).

closed 2020-06-07 by vchrisb