fsantini/python-e3dc #19: WEB RSCP Connection seems not to work

one4many, 2020-05-26:
Hi,
I'm trying to get the websocket based RSCP connection to work (local works fine).
The server seems not to return the desired response:
< SERVER_REGISTER_CONNECTION (virtConId, virtAuthLevel) 
```
python-e3dc alex$ python test.py 
web connection
Decoded received message ('SERVER_REGISTER_CONNECTION', 'Container', [('SERVER_CONNECTION_ID', 'Int64', 43380680), ('SERVER_AUTH_LEVEL', 'UChar8', 60), ('SERVER_TYPE', 'Int32', 1), ('SERVER_HASH_CODE', 'Int32', 71390474)])
DEBUG self.conId, self.authLevel, self.virtConId, self.virtAuthLevel: 43380680 60 None None
b'\xe3\xdc\x00\x11W\x18\xcd^\x00\x00\x00\x00\x83\x02\x00\x00\x1e\x00\x01\xa0\x08\xf8\x0e\x17\x00\x01\xb0\x00\xf8\x08\x08\x00\xc8\xef\x95\x02\x00\x00\x00\x00\x02\xb0\x00\xf8\x03\x01\x00<F\x9c[ '
Decoded received message ('SERVER_REQ_RSCP_CMD', 'Container', [('SERVER_CONNECTION_ID', 'Int64', 43380680), ('SERVER_AUTH_LEVEL', 'UChar8', 60), ('SERVER_RSCP_DATA_LEN', 'Int32', 88), ('SERVER_RSCP_DATA', 'ByteArray', b"\xe3\xdc\x00\x01V\x18\xcd^\x00\x00\x00\x00\x80Qu'F\x00\x08\x00\x00\n\x00\x00\x00\t\x00\x00\n\x00\x00\x00\x0b\x00\x00\n\x00\x00\x00\x0c\x00\x00\n\x00\x00\x00\r\x00\x00\n\x00\x00\x00\x0e\x00\x00\n\x00\x00\x00\x10\x00\x00\n\x00\x00\x00\x0f\x00\x00\n\x00\x00\x00\x07\x00\x00\n\x00\x00\x00\x11\x00\x00\n\x00\x00\x00")])
Inner frame chunk decoded ('INFO_REQ_IP_ADDRESS', 'None', None)
Inner frame chunk decoded ('INFO_REQ_SUBNET_MASK', 'None', None)
Inner frame chunk decoded ('INFO_REQ_GATEWAY', 'None', None)
Inner frame chunk decoded ('INFO_REQ_DNS', 'None', None)
Inner frame chunk decoded ('INFO_REQ_DHCP_STATUS', 'None', None)
Inner frame chunk decoded ('INFO_REQ_TIME', 'None', None)
Inner frame chunk decoded ('INFO_REQ_TIME_ZONE', 'None', None)
Inner frame chunk decoded ('INFO_REQ_UTC_TIME', 'None', None)
Inner frame chunk decoded ('INFO_REQ_A35_SERIAL_NUMBER', 'None', None)
Inner frame chunk decoded ('INFO_REQ_INFO', 'None', None)
Decoded received message ('SERVER_REQ_RSCP_CMD', 'Container', [('SERVER_CONNECTION_ID', 'Int64', 43380680), ('SERVER_AUTH_LEVEL', 'UChar8', 60), ('SERVER_RSCP_DATA_LEN', 'Int32', 67), ('SERVER_RSCP_DATA', 'ByteArray', b'\xe3\xdc\x00\x01V\x18\xcd^\x00\x00\x00\x00@\xbf\xd721\x00\x01\x00\x80\n\r*\x00WEB_9caa9782af8c8b08d4f1c0275a0c2f31414853')])
Inner frame chunk decoded ('INFO_SERIAL_NUMBER', 'CString', 'WEB_9caa9782af8c8b08d4f1c0275a0c2f31414853')
DEBUG OUTPUT: b'**[username.blanked.out]**' **[password.md5.blanked.out]** b'**[serial.blanked.out]**'
--------------------- Sending virtual conn

Here comes no answer from the server ... and the handshake seems to begin from scratch.

Decoded received message ('SERVER_REGISTER_CONNECTION', 'Container', [('SERVER_CONNECTION_ID', 'Int64', 43380807), ('SERVER_AUTH_LEVEL', 'UChar8', 60), ('SERVER_TYPE', 'Int32', 1), ('SERVER_HASH_CODE', 'Int32', 380632551)])
DEBUG self.conId, self.authLevel, self.virtConId, self.virtAuthLevel: 43380807 60 None None
b'\xe3\xdc\x00\x11b\x18\xcd^\x00\x00\x00\x00\x1a\x00\x00\x00\x1e\x00\x01\xa0\x08\xf8\x0e\x17\x00\x01\xb0\x00\xf8\x08\x08\x00G\xf0\x95\x02\x00\x00\x00\x00\x02\xb0\x00\xf8\x03\x01\x00<\x06%}\xcd'
Decoded received message ('SERVER_REQ_RSCP_CMD', 'Container', [('SERVER_CONNECTION_ID', 'Int64', 43380807), ('SERVER_AUTH_LEVEL', 'UChar8', 60), ('SERVER_RSCP_DATA_LEN', 'Int32', 88), ('SERVER_RSCP_DATA', 'ByteArray', b'\xe3\xdc\x00\x01a\x18\xcd^\x00\x00\x00\x00@\xa5\xae\x02F\x00\x08\x00\x00\n\x00\x00\x00\t\x00\x00\n\x00\x00\x00\x0b\x00\x00\n\x00\x00\x00\x0c\x00\x00\n\x00\x00\x00\r\x00\x00\n\x00\x00\x00\x0e\x00\x00\n\x00\x00\x00\x10\x00\x00\n\x00\x00\x00\x0f\x00\x00\n\x00\x00\x00\x07\x00\x00\n\x00\x00\x00\x11\x00\x00\n\x00\x00\x00')])
Inner frame chunk decoded ('INFO_REQ_IP_ADDRESS', 'None', None)
Inner frame chunk decoded ('INFO_REQ_SUBNET_MASK', 'None', None)
Inner frame chunk decoded ('INFO_REQ_GATEWAY', 'None', None)
Inner frame chunk decoded ('INFO_REQ_DNS', 'None', None)
Inner frame chunk decoded ('INFO_REQ_DHCP_STATUS', 'None', None)
Inner frame chunk decoded ('INFO_REQ_TIME', 'None', None)
Inner frame chunk decoded ('INFO_REQ_TIME_ZONE', 'None', None)
Inner frame chunk decoded ('INFO_REQ_UTC_TIME', 'None', None)
Inner frame chunk decoded ('INFO_REQ_A35_SERIAL_NUMBER', 'None', None)
Inner frame chunk decoded ('INFO_REQ_INFO', 'None', None)
Decoded received message ('SERVER_REQ_RSCP_CMD', 'Container', [('SERVER_CONNECTION_ID', 'Int64', 43380807), ('SERVER_AUTH_LEVEL', 'UChar8', 60), ('SERVER_RSCP_DATA_LEN', 'Int32', 67), ('SERVER_RSCP_DATA', 'ByteArray', b'\xe3\xdc\x00\x01a\x18\xcd^\x00\x00\x00\x00@\xf2\x80\x0b1\x00\x01\x00\x80\n\r*\x00WEB_f9ae7a9522d7941e69443c2f389f6791425193')])
Inner frame chunk decoded ('INFO_SERIAL_NUMBER', 'CString', 'WEB_f9ae7a9522d7941e69443c2f389f6791425193')
DEBUG OUTPUT:  b'**[username.blanked.out]**' **[password.md5.blanked.out]** b'**[serial.blanked.out]**'
--------------------- Sending virtual conn
```

Any help appreciated.

Thank you!

fsantini, 2020-05-26:
This was a problem that I thought I solved some time ago... Just to make sure, did you pull the latest commits?

one4many, 2020-05-26:
Yes.
The last commit I see is: 90369bd6798f7eaea0722928c6ed726e7cfe7747
You made some changes to the web stuff ... and that is the version I run.

one4many, 2020-06-02:
Any update on this issue?

fsantini, 2020-06-06:
Sorry I can't reproduce. Does the local connection work? Can you try?

one4many, 2020-06-06:
Yes the local connection works flawless.

is00709, 2020-06-11:
Hi,

thank you for your great work! The local connection is running....

But I have the following problem with the web connection. I activated the websocket trace in "_e3dc_rscp_web.py". This is the response from the E3DC web server:

-----------------------
--- response header ---
HTTP/1.1 101 Switching Protocols
Server: nginx/1.14.2
Date: Thu, 11 Jun 2020 13:10:37 GMT
Connection: upgrade
Sec-WebSocket-Accept: D0bH4iGd1kSF5shm9wdbUwqr1RQ=
Upgrade: websocket
-----------------------
send: b'\x82\xb4\xcb]\x9d\xdd(\x81\x9d\xcc\x85p\x7f\x83\xcb]\x9d\xddh_\x9d\xdd\xd5]\x9c}\xc3\xa5\x93\xca\xcb\\-\xdd3U\x95\xdd\x97\xe8)\xdd\xcb]\x9d\xdd\xc9\xed\x9d%\xc8\\\x9d\xe1\xa6\x8f\xb9g'
send: b'\x82\xfe\x018,\x9fX\x94\xcfCX\x85b\xb2\xba\xca,\x9fX\x94\x0f\x9cX\x94\x0e\x9e[4,gV\x8f-\x9e\xe8\x94\xd4\x97P\x94p*\xec\x94,\x9fX\x94./Xl/\x9eX\xa8(/Xl*\x9bXf,\x9fX\x91\x9c\x9f\xa0\x84\xde\x9f\xbbH,\x8e\x16\xb9\xce\xc1X\x94,\x9f{\x97,\x9f\x84\x94$\x9f\xd8\x9e!\x98X\xa4\x02\xafv\xa4\x02\xafQ\x94\xac\x95U\x93,\xafv\xa4\x02\xafv\xa4\'\x9f\xd8\x9e!\x98X\xa4\x02\xafv\xa4\x02\xafT\x94\xac\x95U\x93,\xafv\xa4\x02\xafv\xa4!\x9f\xd8\x9e-\x9eX\x95"\x9f\xd8\x9e<\x93X\x94,\xdf\x0b\x1f\x94H\x19Y.\x9fX\x84,\x1fR\x99)\x9f\x1f\xd9x\xb4j\x9b,\x1fR\x84 \x9fX\x94l\xd4\xdc,\xfb\xde\x96\x96,\x9f_\x94\xac\x95U\x92,\xaej\xa7\x18\xaan\x85,\x1fR\x9az\x9fY\x94\xac\x95U\xb0,\xc8\x1d\xd6s\xfem\xad\x19\xfbn\xf6M\xfcn\xa2\x1e\xfaa\xa4\x1c\xfe<\xa7N\xfci\xf6\x1a\xfcn\xf5\x14\xa9i\xf2\x1b\x9dX\x14&\x92T\x94\x19\xa8h\xa0\x1d\xad`\xa4\x1c\xafh\xa4&\x9f\xd8\x9e!\x8eX\xa4\x1c\xa5h\xa4\x16\xafh\xae\x1c\xafb\xa4\x1c\xa5h\xa4\xa8\xdc\xa2:\xedR\x1c\xc2'
send: b'\x82\xfe\x00\x8b\xf90\xf1#\x1a\xec\xf12\xb7\x1d\x13}\xf90\xf1#\xb03\xf1#\x8c0\xf4\x83\xf9\xc8\xffM\xf98A#\x01=\xe2#\x98B\x9cJ\x97\x1e\x94A\x9cB\x9dF\xb9G\x94A\xd7T\x94*I0\t.\xd90\xc4G\xce\x04\x97\x17\x9d\x03\xc6F\xc1\x01\x94F\xcd\x05\x92B\xce\x08\xc6G\xc0\x03\xc9\x16\xcf\t\x95\x16\xcfQ\xe1\x93\xf9\xc8\xfc3\xf9c\xc0\x13\xd4\x07\xc1\x12\xc0\x03\xc2\x13\xc9\x00\xc6\x1a\xce6A#\x016\xf5#\xfd0\xf1#\xfe\x80\xf1\xdb\xff4\xf1\xf1\xfb\xa6\xb8\x8a\t\xca\x85'
error from callback <function E3DC_RSCP_web.connect.<locals>.<lambda> at 0xffff96945200>: 176
  File "/usr/local/lib/python3.7/dist-packages/websocket_client-0.57.0-py3.7.egg/websocket/_app.py", line 346, in _callback
    callback(self, *args)
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_e3dc_rscp_web.py", line 267, in <lambda>
    on_message = lambda ws, msg: self.on_message(msg),
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_e3dc_rscp_web.py", line 163, in on_message
    decodedMsg = rscpLib.rscpDecode(message)[0]
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_rscpLib.py", line 153, in rscpDecode
    return rscpDecode(rscpFrameDecode(data)[0])
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_rscpLib.py", line 166, in rscpDecode
    innerData, usedLength = rscpDecode(data[curByte:])
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_rscpLib.py", line 159, in rscpDecode
    strType = rscpTags.getDatatype(hexType)
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_rscpTags.py", line 1941, in getDatatype
    return rscpDataTypes[type_hex]
Traceback (most recent call last):
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_e3dc.py", line 340, in sendRequest
    self.rscp.connect()
  File "/home/e3dc/E3DC/python-e3dc-python3/e3dc/_e3dc_rscp_web.py", line 278, in connect
    if not self.isConnected(): raise RequestTimeoutError
e3dc._e3dc_rscp_web.RequestTimeoutError

Any ideas??

one4many, 2020-06-18:
Hi there,
so trying to debug stuff myself I noted that using the reference accounts E3DC has on their website, User: referenz / Pass: referenz / one of serials: 411431000294 the code actually works. The only notable difference here would be that this user or password does not have 'special' characters. My username and password has dots and dashes in them.

vchrisb, 2020-07-01:
Hey @one4many,

There is a bug in the serial number handling:
https://github.com/fsantini/python-e3dc/blob/master/e3dc/_e3dc.py#L112-L113

`serialNumber` and `serialNumberPrefix` are mixed up.
It probably should be:
```
self.serialNumber = serial[4:]
self.serialNumberPrefix = serial[:4]
```

I'm going to fix this.

one4many, 2020-07-01:
Oh yeah .. my bad, sorry.

vchrisb, 2020-07-01:
no worries. It is fixed. 👍

PR #22 (2020-06-18): Fix serial number handling for rscp requests via web
This fixes  #19 "WEB RSCP Connection seems not to work"
It it a matter of having the correct serial incl. prefix. Which is not always S10- for newer systems.
closed 2020-07-01 by vchrisb