fsantini/python-e3dc #31: [Question] Clarification of PM_INDEX

spali, 2020-11-04:
Hi 
currently discovering your great API.
In my case I get no data on `get_power_data` and found out using pmIndex = 1 works.
I have a Quattroporte.
At least I can explain the pvi data error "NOT_AVAILABLE" (made a pull request)... that is because a Quattroporte can't do anything on the DC side .... so no PV stuff.
Not sure if this should be directly catched at the API for a Quattroporte. Maybe not 🤔 
But The PV generated electricity is measured after converted to AC. And this data is what I get in PM_INDEX 1.

But I'm not sure if changing the pmIndex in general for Quattroporte would be the right way.
I have the feeling the available index numbers are highly dependent on the specific system installation.
And could even be multiple indexes at the same time ?!

I bet somehow the available devices can somehow be discovered. But to prove that we need some tests on different installations. 

For example I found ```EMS_REQ_EXT_SRC_AVAILABLE``` which returned:
```('EMS_EXT_SRC_AVAILABLE', 'Char8', 1)```
Regarding the description in the official Excel, it doesn't sound like to be the PM_INDEX (even it matched in my case). I'm more into that this is kind of "boolean" value. Or maybe (think of the PM_INDEX are somehow hard coded for specific devices) this would return 1 for the device at index 1. But this is just guessing.

At least this is a way to find something out about the specific installation.
Still exploring a bit all the available tags and data... but wanted to ask what you think about this.

PS: pull request for correcting the serialnumber prefix for Quattroporte is in work :D

spali, 2020-11-05:
digged a bit more into this.
tested index 0 to 100 ...
result is, I get the 2 indexes with reasonable values (also available on the local web gui).
index 1 contains the generated solar power on the AC side of the converter.
index 6 contains the "root-counter" of the electricity supplier what corresponds with the grid value in poll().
till index 7 I get an RSCP_ERR_NOT_AVAILABLE (except the two mentioned).
index >= 8 does not report an error on the DEVICE_STATE, but reports RSCP_ERR_OUT_OF_BOUNDS for all values.
So I assume max. supported index is 7.

So multiple questions to what I should do:

1. dependent of what power_data is representing for other types of system, I would suggest to either set pmIndex to 1 or 6 for Quattroporte.
If power_data shows your grid power, then it would be 6. if power_data shows the solar power production then it would be 1.
Could you please give me feedback on this, then I would do a pull request for it?

2. Also I would add EMS_REQ_POWER_ADD to poll() under production key. Suggestion as "add". And accumulate it in the home 
calculation. Agree?

3. Not sure if I'm to perfectionistic. But I could imagin that additional indexes could be also exist for other device types. So what should be do for this? I couldn't find a way to query the existing indexes. So I think a try and error approach (index 0-7) is the only way. I also suggest (even it's a breaking change) to return the existing indexes either as array or object with index/key = device index for the power_data. What do you think?

spali, 2020-11-05:
More thinking...
Ok ... traced the data of the GUI. So correct me if I'm wrong... but power_data shows the grid power in your case. So I would use index 6 for Quattroporte.
From this point and regarding EMS_REQ_EXT_SRC_AVAILABLE could be interpreted as that only one additional index to the grid index could be valid.

If you agree, then we could solve it like this.
we could just add this info to system_info with the key "additionalPowerMeterInstalled"
and create a method power_data_add which by default uses index 1 for all devices.
then we do not have a breaking change, but must hope, that not more than one additional power meter can exist.

fsantini, 2020-11-09:
Thanks for all your research! You discovered a lot more than I knew. I'll check your PR out.

fsantini, 2020-11-09:
I noticed that you added the "add" in the production side, but then you subtract this value from the consumption calculation. Is this intended? In principle, "production" values should be positive when energy flows in the house, which is opposite of the calculation that you make. Is there a reason for this?

spali, 2020-11-09:
Good point, yes it was intendet, in my case it was a negativ value for the "production", so I had to subtract it to get the balance right.
But maybe this is depending on how the installation is implemented. Any idea how we should proceed here?

fsantini, 2020-11-09:
I think that it must be standardized; so either you count it among the "consumption" or, if it makes more sense as a "source", you should return the value `- add` in the outObj

PR #33 (2020-11-12): Quattroporte changes
suggested solution for Quattroporte 
Fixes #31
closed 2020-11-12 by fsantini