fsantini/python-e3dc #37: Connection is not working

jhubig, 2020-11-22:
Hello,

I'm not able to connect to my E3DC. Tried the local and also web connection. Both are failing.

```bash
pi@raspberry3:~ $ python3 .homebridge/Cmd4Scripts/bat_connect.py
web connection
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 386, in sendRequest
    self.rscp.connect()
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc_rscp_web.py", line 278, in connect
    if not self.isConnected(): raise RequestTimeoutError
e3dc._e3dc_rscp_web.RequestTimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File ".homebridge/Cmd4Scripts/bat_connect.py", line 11, in <module>
    e3dc = E3DC(E3DC.CONNECT_WEB, username=USERNAME, password=PASS, serialNumber = SERIALNUMBER, isPasswordMd5=False)
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 112, in __init__
    self.get_system_info_static(keepAlive=True)
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 645, in get_system_info_static
    self.deratePercent  = round(self.sendRequest( ('EMS_REQ_DERATE_AT_PERCENT_VALUE', 'None', None), keepAlive = True  )[2] * 100)
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 396, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
```

Here the output after trying the local connection:

```bash
pi@raspberry3:~ $ python3 .homebridge/Cmd4Scripts/bat_connect.py
local connection
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc_rscp_local.py", line 53, in sendRequest
    self._send(plainMsg)
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc_rscp_local.py", line 40, in _send
    encData = self.encdec.encrypt(sendData)
AttributeError: 'NoneType' object has no attribute 'encrypt'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 387, in sendRequest
    result = self.rscp.sendRequest( request )
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc_rscp_local.py", line 57, in sendRequest
    raise CommunicationError
e3dc._e3dc_rscp_local.CommunicationError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File ".homebridge/Cmd4Scripts/bat_connect.py", line 19, in <module>
    e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress = TCP_IP, key = KEY)
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 112, in __init__
    self.get_system_info_static(keepAlive=True)
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 645, in get_system_info_static
    self.deratePercent  = round(self.sendRequest( ('EMS_REQ_DERATE_AT_PERCENT_VALUE', 'None', None), keepAlive = True  )[2] * 100)
  File "/usr/local/lib/python3.5/dist-packages/e3dc/_e3dc.py", line 396, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
pi@raspberry3:~ $
```

Here my python script. I have checked all needed input parameters several times. They should be okay. Thanks for your help.

```python
#!/usr/bin/python3
from e3dc import E3DC

TCP_IP = 'xxx'
USERNAME = 'xxx'
PASS = 'xxx'
KEY = 'xxx'
SERIALNUMBER = 'xxx'
#
print("web connection")
e3dc = E3DC(E3DC.CONNECT_WEB, username=USERNAME, password=PASS, serialNumber = SERIALNUMBER, isPasswordMd5=False)
# connect to the portal and poll the status. This might raise an exception in case of failed login. This operation is performed with Ajax
print(e3dc.poll())
# Poll the status of the switches using a remote RSCP connection via websockets
# return value is in the format {'id': switchID, 'type': switchType, 'name': switchName, 'status': switchStatus}
print(e3dc.poll_switches())

print("local connection")
e3dc = E3DC(E3DC.CONNECT_LOCAL, username=USERNAME, password=PASS, ipAddress = TCP_IP, key = KEY)
# The following connections are performed through the RSCP interface
print(e3dc.poll())
print(e3dc.poll_switches())
```

Cheers,
Johannes

jhubig, 2020-11-23:
I'm so stupid. I misunderstood the KEY value. Thought it is the Cloud Registry Key from the E3DC Portal. In fact it is the RSCP-Key entered on your E3DC battery in the SETTINGS -> PERSONALIZE menu. Now my local connection is working.

Web connection still not working, but the local should be enough for the moment.

TCP_IP = '<IP ADDRESS>'
USERNAME = '<USER from E3DC Portal login>'
PASS = '<PASSWORD from E3DC Portal login>'
KEY = '<Value entered in E3DC in SETTINGS -> PERSONALIZE menu for RSCP-Key>'

Cheers,
Johannes

closed 2020-11-23 by jhubig

danielgauss, 2021-01-10:
Hello,

I realize this has been closed, though the web connection issue still seems to be unresolved.
In the meantime, I tried it out on a linux and on a windows system, but the result is the same on both:
```
Traceback (most recent call last):
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python39\lib\site-packages\e3dc-0.4.0-py3.9.egg\e3dc\_e3dc.py", line 386, in sendRequest
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python39\lib\site-packages\e3dc-0.4.0-py3.9.egg\e3dc\_e3dc_rscp_web.py", line 278, in connect
e3dc._e3dc_rscp_web.RequestTimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "E:\python\pye3dc", line 18, in <module>
    web = E3DC(E3DC.CONNECT_WEB, username=e3user, password=keyforweb, serialNumber=serialnr, isPasswordMd5=False)
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python39\lib\site-packages\e3dc-0.4.0-py3.9.egg\e3dc\_e3dc.py", line 112, in __init__
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python39\lib\site-packages\e3dc-0.4.0-py3.9.egg\e3dc\_e3dc.py", line 645, in get_system_info_static
  File "C:\Users\daniel\AppData\Local\Programs\Python\Python39\lib\site-packages\e3dc-0.4.0-py3.9.egg\e3dc\_e3dc.py", line 396, in sendRequest
e3dc._e3dc.SendError: Max retries reached
```

I double-checked the credentials in my script. Does anyone else have this connection issue? Or does anybody know how to resolve it? I'm using the latest revision of the repository as of today. (2021-01-10)

Regards
Daniel

fsantini, 2021-01-11:
Hi. Thanks. Unfortunately, I can't reproduce the problem because it works for me. Maybe it is indeed a timeout error. Does it happen as soon as you try to create the E3DC object?

danielgauss, 2021-01-12:
Hi again. with my limited knowledge of python I think it happens when the E3DC object is created, though I'm not 100% sure.
When I find the time I'll try to add some more debugging output, maybe it's possible to get a better understanding of what exactly is failing.

Regards
Daniel

skuske, 2021-09-01:
Web connection still not working for me, even with tzlocal 2.1

```
web connection
Traceback (most recent call last):
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 386, in sendRequest
    self.rscp.connect()
  File "C:\python\lib\site-packages\e3dc\_e3dc_rscp_web.py", line 278, in connect
    if not self.isConnected(): raise RequestTimeoutError
e3dc._e3dc_rscp_web.RequestTimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\python\test.py", line 8, in <module>
    e3dc = E3DC(E3DC.CONNECT_WEB, username=USERNAME, password=PASS, serialNumber = SERIALNUMBER, isPasswordMd5=False)
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 112, in __init__
    self.get_system_info_static(keepAlive=True)
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 645, in get_system_info_static
    self.deratePercent  = round(self.sendRequest( ('EMS_REQ_DERATE_AT_PERCENT_VALUE', 'None', None), keepAlive = True  )[2] * 100)
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 396, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
```

Any solution?

Many thanks!