fsantini/python-e3dc #38: type annotation requires Python 3.6+

vchrisb, 2020-11-23:
in https://github.com/fsantini/python-e3dc/blob/master/e3dc/_e3dc.py#L603 Python type annotation is used, which is only supported since Python 3.6 https://www.python.org/dev/peps/pep-0526/

We should either remove this type annotation or document a requirement for Python 3.6+.

What do you think @fsantini

fsantini, 2020-11-23:
I'm not a big fan of type annotations in Python. My position is: if you want a statically-typed language, write in C++ :)
Since this is introducing backwards incompatibilities I would remove them. We should also consider that many users will run this on a raspberry pi that might not be updated in a while (that's what I do actually).

From what I've seen, it doesn't seem that you can import those from `__future__`, so I would privilege compatibility over this (in my eyes) minor syntactic sugar.

fsantini, 2020-11-23:
Also, the rest of the library doesn't use them.

closed 2021-11-30 by vchrisb