fsantini/python-e3dc #39: PVI Data werden falsch abgerufen

rxhan, 2020-11-30:
Hallo,

die PVI-Daten werden falsch abgerufen. Den folgenden Parametern wird richtigerweise die abzufragene Phase mitübergeben: PVI_REQ_AC_POWER, PVI_REQ_AC_VOLTAGE, PVI_REQ_AC_CURRENT, PVI_REQ_AC_APPARENTPOWER, PVI_REQ_AC_REACTIVEPOWER, PVI_REQ_AC_ENERGY_ALL, PVI_REQ_AC_ENERGY_GRID_CONSUMPTION (also 0-2 bei 3 Phasigen Wechselrichtern, abfragbar mit: PVI_AC_MAX_PHASE_COUNT)

Die folgenden Tags müssen je String abgerufen werden: PVI_REQ_DC_POWER, PVI_REQ_DC_VOLTAGE, PVI_REQ_DC_CURRENT, PVI_REQ_DC_STRING_ENERGY_ALL  (0-1 bei 2 Strings, abfragbar mit: PVI_DC_MAX_STRING_COUNT)

Die Temperaturen sind abhängig der Anzahl in PVI_TEMPERATURE_COUNT abzufragen: PVI_REQ_TEMPERATURE

Aktuell unterscheidet die Funktion get_pvi_data nicht dazwischen, sondern ruft für alle Tags die selbe Nummer ab. Bei pviTracker = 0 wird dann die Spannung von Phase 1, die Spannung von Tracker 1 und die erste Temperatur geliefert. Bei pviTracker=2 wird ein Fehler produziert, die dritte Phase ist damit nicht abrufbar.

Gemeint ist der Teil ab hier:
https://github.com/fsantini/python-e3dc/blob/f0c7755c936347f155f7e7e8cd093396c5536f83/e3dc/_e3dc.py#L810

vchrisb, 2020-11-30:
Danke. Ich werde in English antworten, da nicht jeder Nutzer dieser Library Deutsch spricht. Falls das für dich wiederum schwierig ist, dann bitte das erwähnen. 👍 

Thank you. I'll answer in english to give non-german speakers the chance to understand the issue. 

So think I understood the issue here. I also looked at your implementation: https://github.com/rxhan/RSCPGui/blob/d39dae9d5ed72ae3214630b4a2f11ac4557b728e/e3dc/rscp_helper.py#L408-L462

One challenge is, that there a quite a few different E3DC models and configuration options and e.g. myself only knows my setup.
Am I right, that a E3DC model can have multiple inverter (PVI_INDEX) where each of these converters can have one or more strings (DC side) and on or more phases (AC side). Furthermore each inverter can provide multiple temperature values?

That would mean that we mixed `stringIndex` and `pviTracker` + not gathering all the correct data. :(

rxhan, 2020-11-30:
You are correct in your assumptions, my implementation has been tested by many different systems. Unfortunately not yet with several inverters and not even with a Quadroporte.

spali, 2021-02-01:
Hi
I contributed already some Quadroporte fixes. during my past experience I got the feeling, that it's not only different per type, but also dependent on the specific on site implementation. i.e. in my situation, the `PVI_*` stuff doesn't work completely, my solar production is implemented as an additional "general" inverter that just is counted as a positive value.

That is why I switched to my own library which allows to call with "raw" tags whatever you want and does not rely on the system specific stuff for which it is always never possible to keep up to date and handle every use-case.
Don't understand me wrong, no offend to this excellent library, used it quite a while and it works for the most people.
But I wanted also to get some other values and also have more "write" access to the system with need to implement that specifically. So I just implemented a command line utility that speaks RSCP ;)

PR #56 (2022-02-21): multiple enhancements
* adds and fixes docstrings in complete project, adresses #55 
* adds docstring flake8 tests to github action
* removes unused `_e3dc_rscp.py`
* fixes flake8 extend-ignore configuration
* enhances `get_battery_data` by adding dcb data, fixed #41
* fixes and enhances  `get_pvi_data`, fixes #39
* enhances `get_power_data`
* fixes `E741` and `F601` flake8 test
* `configuration` object for E3DC class to set system specific configuration, like multiple batteries or index of PM
* added `get_pvis_data`, `get_powermeters_data` and `get_batteries_data`
* added simplified functions [... cut]
closed 2022-02-21 by vchrisb