fsantini/python-e3dc #43: Web connection not working

skuske, 2021-09-01:
Double checked username, password end serial number, but no chance.

The errors I get:

```
web connection
Traceback (most recent call last):
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 386, in sendRequest
    self.rscp.connect()
  File "C:\python\lib\site-packages\e3dc\_e3dc_rscp_web.py", line 278, in connect
    if not self.isConnected(): raise RequestTimeoutError
e3dc._e3dc_rscp_web.RequestTimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\python\test.py", line 8, in <module>
    e3dc = E3DC(E3DC.CONNECT_WEB, username=USERNAME, password=PASS, serialNumber = SERIALNUMBER, isPasswordMd5=False)
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 112, in __init__
    self.get_system_info_static(keepAlive=True)
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 645, in get_system_info_static
    self.deratePercent  = round(self.sendRequest( ('EMS_REQ_DERATE_AT_PERCENT_VALUE', 'None', None), keepAlive = True  )[2] * 100)
  File "C:\python\lib\site-packages\e3dc\_e3dc.py", line 396, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
```

skuske, 2021-09-01:
OK, I think it might be related to the fact the S10E serial numbers might start with a 7.

My S10E does.

bonsaischaf, 2022-04-08:
It looks like for my case I got it running with my S10 E Compact unit by modifying two things:

- as you stated above, the serial number starts with 74 instead of 4. The modification in _set_serial(...) did the first part of the job (S10 E -> S10-72..., S10 E Compact -> S10-74 ..., S10 E Pro -> P10-70..., S10 E Compact Pro -> P10-75... according to v6.80 of the manual)
` if self.serialNumber.startswith("4"):`
- for my unit the polled values of my powermeter are indexed with PM1_L* instead of PM0_L*. Changing self.pmIndexExt from 0 to 1 in _set_serial(...) did the second part of the job.
` self.pmIndexExt = 0`

skuske, 2022-04-09:
@bonsaischaf 

Thanks. Can you tell me on which page of the manual you found the information about the serial numbers?

vchrisb, 2022-04-11:
Thank you @bonsaischaf. I wasn't aware of the list of models and its corresponding Serial Numbers. I will update the according code.
According to E3DC, the root power meter is either `0` or `6`. Other indexes are used for additional power meters.
I've added in a recent release the option to optionally configure the pmIndex, and add additional ones, while creating the E3dc Object.

skuske, 2022-04-11:
@vchrisb 

I know that there are also systems with serial numbers starting with 49, 81, 50, 54. Maybe there are even more. But I do not know what type of systems are linked to those. The docs don't say anything about it.

That makes it even more difficult ...

vchrisb, 2022-04-11:
I **think** the following might be correct:
* `49` old S10 E
* `81` S10 X
* `54` S10 Mini
* `50` not sure about this one, but maybe earlier S10 mini models

vchrisb, 2022-05-24:
fixed by #65 and #63

closed 2022-05-24 by vchrisb