fsantini/python-e3dc #50: Parameter timespan in get_db_data does not work

initB10r, 2021-11-22:
I can't see any difference in the result when changing timespan.

Here is my example:

Call 1:
```
myDate = datetime.date(2021, 11, 22)
print(e3dc.get_db_data(myDate,'DAY'))
```
Result 1:
```
{'bat_power_in': 15415.75, 'bat_power_out': 448.75, 'solarProduction': 31404.0, 'grid_power_in': 3985.75, 'grid_power_out': 6728.75, 'consumption': 17748.5, 'stateOfCharge': 84.84100341796875, 'consumed_production': 73.43807220458984, 'autarky': 62.088348388671875}
```

Call 2:
```
myDate = datetime.date(2021, 11, 22)
print(e3dc.get_db_data(myDate,'MONTH'))
```
Result 2:
```
{'bat_power_in': 15415.75, 'bat_power_out': 448.75, 'solarProduction': 31404.0, 'grid_power_in': 3985.75, 'grid_power_out': 6728.75, 'consumption': 17748.5, 'stateOfCharge': 84.84100341796875, 'consumed_production': 73.43807220458984, 'autarky': 62.088348388671875}
```

Call 3:
```
myDate = datetime.date(2021, 11, 22)
print(e3dc.get_db_data(myDate,'YEAR'))
```
Result 3:
```
{'bat_power_in': 15415.75, 'bat_power_out': 448.75, 'solarProduction': 31404.0, 'grid_power_in': 3985.75, 'grid_power_out': 6728.75, 'consumption': 17748.5, 'stateOfCharge': 84.84100341796875, 'consumed_production': 73.43807220458984, 'autarky': 62.088348388671875}
```

I have seen in the code that the container "DB_REQ_HISTORY_DATA_DAY" is always used. What about the containers "TAG_DB_REQ_HISTORY_DATA_WEEK", "TAG_DB_REQ_HISTORY_DATA_MONTH" or "TAG_DB_REQ_HISTORY_DATA_YEAR"?

What exactly is expected in the fields "DB_REQ_HISTORY_TIME_START", "DB_REQ_HISTORY_TIME_INTERVAL" and "DB_REQ_HISTORY_TIME_SPAN"?

vchrisb, 2021-11-22:
You used the current day as the start day. Hence there isn't any data after the 22nd of November yet and hence day, month and year have the same value.
Try using a day in the past as the start date.

initB10r, 2021-11-22:
Wouldn't it be nicer if the programme automatically determined the 1st day of the month of the date entered in MONTH mode? In my example, 01.11.2021 and for the year 01.01.2021.

Otherwise, a monthly value at the query value 15.08.2021 would query until 15.09.2021.

Perhaps a switch to change the behaviour would also be helpful.

vchrisb, 2021-11-23:
Maybe you can do a PR?
There are a lot of areas for improvement and sadly the documentation on the Tags and their usage isn't well, or at all, documented by E3DC. Most stuff is try and error.

vchrisb, 2022-03-06:
Fixed by #57

closed 2022-03-06 by vchrisb