fsantini/python-e3dc #51: get_power_data throws TypeError

mdhom, 2021-11-23:
First of all: MANY THANKS for this great library which made it easy to fetch data from my E3/DC system.
I get the poll data working, now i wanted to fetch the power data:
```
power_data = e3dc.get_power_data()
print(power_data)
```
But it throws an error:
` <class 'TypeError'> 'NoneType' object is not subscriptable`

Am i doing something wrong here?

Thanks,
Max

mdhom, 2021-11-23:
I printed the result of the `sendRequest` call and received this:
```
('PM_DATA', 'Container', [('PM_INDEX', 'Uint16', 0), ('PM_DEVICE_STATE', 'Error', 'RSCP_ERR_NOT_AVAILABLE')])
```

mdhom, 2021-11-23:
Could you give me a link or something where that RSCP protocol is documented? I can't find anything around the internet but would be very willing to contribute here.

vchrisb, 2021-11-23:
what E3DC Model do you have?
Sadly there isn't much documentation available from E3DC.  In the download section of the E3DC portal is only some documentation from 2016...

mdhom, 2021-11-23:
E3/DC S10E Pro is my model

mdhom, 2021-11-23:
> what E3DC Model do you have? Sadly there isn't much documentation available from E3DC. In the download section of the E3DC portal is only some documentation from 2016...

Ah, found those two documents. But where did you get tags like `PM_REQ_POWER_L1` from? Can't find that anywhere in these pdfs...

vchrisb, 2021-11-23:
Reverse engineering and try and error. The portal is also using rscp and has a list of all tags. https://s10.e3dc.com/s10/js/rscpLibV0.9.2.min.js

vchrisb, 2021-11-24:
You can try a different value for `pmIndex` https://github.com/fsantini/python-e3dc/blob/master/e3dc/_e3dc.py#L140. Not sure if someone else validated if `pmIndex = 0` is correct for a S10 Pro.
@fsantini has a S10E and I have a S10mini. So these are the products we test against.

mdhom, 2021-11-27:
@vchrisb, @fsantini added pull request that fixes the pmIndex for S10 Pro, tested it with my system here!

PR #52 (2021-11-28): Fixed pmIndex for S10 Pro
# Situation
Closes #51, `get_power_data` failed on my S10 Pro system.

# Fix
By trying several different values for `pmIndex` I managed to get power data results when using `pmIndex=6`.
closed 2021-11-28 by vchrisb