fsantini/python-e3dc #54: Retrieve Wallbox history data

mdhom, 2022-01-13:
Hey folks,

do you know of any possibility to get the wallbox history data, that can be seen in the online portal?
[image]

Thanks for your help!

Max

mdhom, 2022-01-13:
the portal also seems to have some kind of "db data" for the wallbox:
[image]

mdhom, 2022-03-02:
just fyi, I opened a support ticket at E3DC to ask for help

marius-klemm, 2023-10-15:
@mdhom @vchrisb Is there any news regarding this enhancement? The python module is really great, but unfortunately the wallbox data is missing. I would also like to connect this via the internet.

Has the support of e3dc commented on this and have you found a way to implement the wallbox data? I came across your pull request here, but it doesn't seem to be remembered yet, does it?

Thank you guys already for your feedback as well as in general for the work you each have done with here already.

Many greetings
Marius

derhappy, 2023-11-24:
I don't know any other implementation supporting the WB history data in this detail. This information was added to the E3DC website long after the "documentation" of the RSCP protocol, so there is no information on how to retrieve this data (if it was added to the protocol at all).
However, it is possible to the the total WB consumption (total, net and solar) with a REQ_HISTORY_* request.
For Example

derhappy, 2023-11-29:
I wanted to start working on implementing some initial wallbox features, but quickly realized @mdhom already did all the work in PR https://github.com/fsantini/python-e3dc/pull/59. Unfortunately it's not merged for over a year. Can I support you in any way Max?

mad61, 2024-05-31:
> I don't know any other implementation supporting the WB history data in this detail. This information was added to the E3DC website long after the "documentation" of the RSCP protocol, so there is no information on how to retrieve this data (if it was added to the protocol at all).
> 
> However, it is possible to get the total WB consumption (total, net and solar) using a REQ_HISTORY_* request. The following request would fetch the WB data for a day, in addition to the other data that's already implemented in this library.
> 
> ```
> [
>   [ # day current
>     "DB_REQ_HISTORY_DATA_DAY",
>     [
>       ["DB_REQ_HISTORY_TIME_START", sometimestamp ],
>       ["DB_REQ_HISTORY_TIME_INTERVAL", sometimestamp],
>       ["DB_REQ_HISTORY_TIME_SPAN", one day ]
>     ]
>   ],
>   [
>     "WB_REQ_DATA",
>     [
>       ["WB_INDEX", 0],
>       "WB_REQ_ENERGY_ALL",
>       "WB_REQ_ENERGY_SOLAR"
>     ]
>   ]
> ]
> ```

I tried to get wallbox history data this way, but without success.

```python
response = self.sendRequest(
                    (
                        RscpTag.DB_REQ_HISTORY_DATA_DAY,
                        RscpType.Container,
                        [
                            (
                                RscpTag.DB_REQ_HISTORY_TIME_START,
                                RscpType.Uint64,
                                startTimestamp,
                            ),
                            (
                                RscpTag.DB_REQ_HISTORY_TIME_INTERVAL,
                                RscpType.Uint64,
                                timespanSeconds,
                            ),
                            (
                                RscpTag.DB_REQ_HISTORY_TIME_SPAN,
                                RscpType.Uint64,
                                timespanSeconds,
                            ),
                        ],
                    ),
                    (
                        RscpTag.WB_REQ_DATA,
                        RscpType.Container,
                        [
                            (
                                RscpTag.WB_INDEX,
                                RscpType.Uint32,
                                0,
                            ),
                            RscpTag.WB_REQ_ENERGY_ALL,
                            RscpTag.WB_REQ_ENERGY_SOLAR,
                        ],
                    ),
            keepAlive=keepAlive,
        )
```

Response always contains just the data like ``def get_db_data_timestamp```.

 Has anyone an idea how to get the history data from wallbox ?