fsantini/python-e3dc #58: SET_WALLBOX_START and SET_WALLBOX_STOP?

othorg, 2022-03-06:
Hi,

do you think, it's possible to implement a simple SET function for the E3DC Wallbox smart connect for starting and stopping the charging-process?

In the S10 Webportal can i see in the wallbox.js script this code for stopping charging:
`var barry = new Uint8Array(6);
		barry[0] = 0;barry[1] = 0;barry[2] = 0;barry[3] = 0;barry[4] = 1;barry[5] = 0;  
                      return dataProxy.rscp.request(
			dataProxy.rscp.containerFactory("WB_REQ_DATA",
				dataProxy.rscp.dataFactory("WB_INDEX","UChar8",currentWallBox),
				dataProxy.rscp.containerFactory("WB_REQ_SET_EXTERN",
					dataProxy.rscp.dataFactory("WB_EXTERN_DATA","ByteArray",barry),
					dataProxy.rscp.dataFactory("WB_EXTERN_DATA_LEN","UChar8",6)`

and for starting:

`var barry = new Uint8Array(6);
		barry[0] = 0;barry[1] = 0;barry[2] = 0;barry[3] = value;barry[4] = 0;barry[5] = 0;
		return dataProxy.rscp.request(
			dataProxy.rscp.containerFactory("WB_REQ_DATA",
				dataProxy.rscp.dataFactory("WB_INDEX","UChar8",parseInt(currentWallBox)),
				dataProxy.rscp.containerFactory("WB_REQ_SET_EXTERN",
					dataProxy.rscp.dataFactory("WB_EXTERN_DATA","ByteArray",barry),
					dataProxy.rscp.dataFactory("WB_EXTERN_DATA_LEN","UChar8",6)`

Or do you think, it's possible to implement the whole function of the wallbox.js?

Thanks in advance

vchrisb, 2022-03-06:
This should be possible, but not using `wallbox.js`, but rather implementing it similar to other methods in `_e3dc.py`.
e.g. https://github.com/fsantini/python-e3dc/blob/master/e3dc/_e3dc.py#L1840

The tags can be found e.g. here: https://github.com/fsantini/python-e3dc/blob/master/e3dc/_rscpTags.py#L1244
and testing can be done using `e3dc.sendRequest()`
 
I don't have a wallbox and hence can't test it.
There are other folks active in this project with a wallbox that might be open to collaborate.

mdhom, 2022-03-06:
Sounds interesting, I could have look at it maybe tonight.

mdhom, 2022-03-06:
First I implemented the `WB_REQ_DATA` to see if I can actually see the live wallbox data, that works. Then I implemented the method which is called `setStopCharging` in `wallbox.js`, but that actually looks more like a toggle charging functionality, which matches my observation that there is no `setStartCharging` function in JS. It's a pity that the bytearray I have to send is just magic, I don't know if there could be any other byte-array I could send to definitely start or stop the wallbox.

@othorg you could give it a try with my current branch here:
https://github.com/mdhom/python-e3dc/tree/wallbox-rscp

Things I need / want to do:

- [ ] Read current state of sun mode
- [ ] Read current state of one phased charging

mdhom, 2022-03-06:
@othorg the code you posted for starting charging seems to be for switching between one and three phased charging

othorg, 2022-03-06:
Hi, 
you're fast ;-)

Funny.... My code is near the same... i played the last two hours...

```
def set_sun_mode(self, enable, keepAlive=False):
            """Setting the Wallbox Sun mode function via rscp protocol locally.

            Args:
                enable (bool): True/False
                keepAlive (Optional[bool]): True to keep connection alive

            Returns:
                0 if success
                -1 if error
            """
            
            barry_on = bytearray([1,0,0,0,0,0])
            barry_off = bytearray([2,0,0,0,0,0])

            if enable:
                #pprint(barry_on)
                res = self.sendRequest(
                    (
                        "WB_REQ_DATA",
                        "Container",
                        [
                            ("WB_INDEX", "UChar8", 0),
                            ("WB_REQ_SET_EXTERN", "Container",
                            [
                                ("WB_EXTERN_DATA", "ByteArray", barry_on), 
                                ("WB_EXTERN_DATA_LEN", "UChar8", 6),
                            ])
                        ],
                    ),
                    keepAlive=keepAlive,
                )
                pprint(res)
            
            else:
                 res = self.sendRequest(
                    (
                        "WB_REQ_DATA",
                        "Container",
                        [
                            ("WB_INDEX", "UChar8", 0),
                            ("WB_REQ_SET_EXTERN", "Container",
                            [
                                ("WB_EXTERN_DATA", "ByteArray", barry_off), 
                                ("WB_EXTERN_DATA_LEN", "UChar8", 6),
                            ])
                        ],
                    ),
                        keepAlive=keepAlive,
                )
            pprint(res)
```

othorg, 2022-03-06:
I will connect my car again asap. After this, i will inspect the start/stop Button in Webportal. my function above is working and switch the sun mode on/off.

mdhom, 2022-03-06:
yes, setting sunmode works perfectly, is also implemented in my code. just starting and stopping charging is somehow strange...

mdhom, 2022-03-06:
do you have a Schuko on your wallbox to test that behavior? I implemented reading out its state and setting it too, but nothing happens here, I think because I don't have a Schuko on my wallbox

othorg, 2022-03-06:
No Sorry i don't have a Schuko... but E3DC have two kinds of WB's. What do you have?
I have the newer one "Smart Connect" 22KW with 3Phases and fixed cable... And for me it looks like, that the start behavior is a toggle function of setStopCharging()?

```
self.stopCharging = function (){
		if (window.OWNER !==3) {
      language.addModule(24).then(function(){
          AXConfirm(language.gE('346M24'),function yes(){
            setStopCharging();
          }, function no(){
            // do nothing
          });
      });	
			
		} 
	};
	self.restartCharging = function (){
		if (window.OWNER !==3) setStopCharging();
		
	};
	
	
}

function setStopCharging(){
	if (window.OWNER ===3) {
		return;
	}
	if (!inUpdate) {
		var barry = new Uint8Array(6);
		//barry[0] = 0;barry[1] = 0;barry[2] = 0;barry[3] = 0;barry[4] = 0;barry[5] = 1;
		barry[0] = 0;barry[1] = 0;barry[2] = 0;barry[3] = 0;barry[4] = 1;barry[5] = 0;

	
		return dataProxy.rscp.request(
			dataProxy.rscp.containerFactory("WB_REQ_DATA",
				dataProxy.rscp.dataFactory("WB_INDEX","UChar8",currentWallBox),
				dataProxy.rscp.containerFactory("WB_REQ_SET_EXTERN",
					dataProxy.rscp.dataFactory("WB_EXTERN_DATA","ByteArray",barry),
					dataProxy.rscp.dataFactory("WB_EXTERN_DATA_LEN","UChar8",6)
				)
			)
		).then(function(data){
			// if(typeof data[0].value.WB_REQ_SET_EXTERN !== "undefined" && data[0].value.WB_REQ_SET_EXTERN.type === "Error"){
			// 	return Promise.reject();
			// }
			// else {
			// 	return Promise.resolve();
			// }
		}).catch(function(err){
			console.error(err);
			// return Promise.reject();
		});
	}
	  
```

mdhom, 2022-03-06:
I've got the E3/DC Wallbox Easy connect flex (no fixed cable). So do you also think we should call the method "toggle"?

mdhom, 2022-03-06:
I created a draft PR: #59 
I think we could discuss the implementation details there.

Balu-78, 2023-10-29:
I like what you are planning to implement. 
I was scanning the repo, as I wanted to change the charging priorioty from wallbox to battery and back
on certain conditions... let's say when the battery of the e3dc system reaches let's say 80% set the charging priority to wallbox.
Do yo think this would be achievable as well in general ?

klepptor, 2023-11-12:
Keep up your great work here!

Looking forward to seeing this implemented for all the projects relying on this library as it's necessary to get the most out of dynamic power tariffs.