fsantini/python-e3dc #68: Writing Value for EMS_EP_RESERVE

hismastersvoice, 2022-10-16:
I tried to write the value for Emergency-Power-Reserve via RSCP to my E3DC
Reading the value works fine.

But if I try to write, I always get NULL back and the value don't change.
Can anybody help me please? Thanks in advance. 

I have found that this value is % of the installed batterie capacity, and it is given als float32 eg 25.00 etc.

    def set_epr(self, ep_reserve=None, keepAlive=False):

        res = self.sendRequest(
            (
                "EMS_REQ_EP_RESERVE",
                "Container",
                [
                    ("EMS_EP_RESERVE", "Float32", ep_reserve),
                ],
            ),
            keepAlive=keepAlive,
        )

Selaron, 2022-10-16:
After several attempts I did it like that and it worked, setting EP Reserve to 1650 Wh:

```
from e3dc import E3DC

...

retval = e3dc.rscp.sendRequest(
        (
            "SE_REQ_SET_EP_RESERVE",
            "Container",
            [
                ("SE_PARAM_INDEX", "Uint32", 0),
                ("SE_PARAM_EP_RESERVE", "Float32", 0.0),
                ("SE_PARAM_EP_RESERVE_W", "Float32", 1650.0),
            ],
        )
    );

```

Selaron, 2022-10-21:
Only thing is that for some reason the new value does not update on the touch screen UI, but it actually works.

hismastersvoice, 2022-10-21:
@Selaron 

I can confirm this.
The display stays with the old value.

Also reading the set value was a bit difficult.
You only get a % value, but this % is calculated by installed capacity and state of health of the batteries.

For me it work perfectly now.

Selaron, 2022-10-21:
Using this you can get the Wh reserve setting:

```
retval = e3dc.rscp.sendRequest(
                ("SE_REQ_EP_RESERVE", "None", None),
    )[2][2][2];

```
The `[2][2][2]` extracts the value out of the tuple.