fsantini/python-e3dc #71: get_battery_data:  'NoneType' object cannot be interpreted as an integer

baststar, 2023-04-01:
Get this error when i run e3dc-to-mqtt (https://github.com/mdhom/e3dc-to-mqtt)

File "/home/pi/gitstuff/e3dc-to-mqtt/e3dc_to_mqtt/e3dc_to_mqtt_base.py", line 164, in run
    battery_data = await self.e3dc.get_battery_data()
  File "/home/pi/gitstuff/e3dc-to-mqtt/e3dc_to_mqtt/e3dc_to_mqtt_base.py", line 299, in get_battery_data
    battery_data = self.__e3dc.get_battery_data(batIndex=i)
  File "/root/.local/lib/python3.7/site-packages/e3dc/_e3dc.py", line 1324, in get_battery_data
    for sensor in range(0, sensorCount):
TypeError: 'NoneType' object cannot be interpreted as an integer

baststar, 2023-09-24:
Paste the code and the error to ChatGPT:

The error message you provided indicates that there's an issue in the `_e3dc.py` file, on line 1324, where the code is attempting to iterate over a range with `sensorCount` as the upper limit. However, it seems that `sensorCount` is `None`, hence the error.

To help you further, I will inspect the `_e3dc.py` file, specifically around line 1324, to identify the source of the problem and suggest a potential solution.

From the code snippet, the error occurs at this line:

```python
for sensor in range(0, sensorCount):
```

Here, `sensorCount` is expected to be an integer, but it's returning `None`. The value of `sensorCount` is determined by the line:

```python
sensorCount = rscpFindTagIndex(info, "BAT_DCB_NR_SENSOR")
```

It seems that the function `rscpFindTagIndex` might be returning `None` when it doesn't find the specified tag within the provided data (`info` in this case).

To resolve this issue, you could:

1. Ensure that the tag "BAT_DCB_NR_SENSOR" is present in the data.
2. Add a check to handle the case when `sensorCount` is `None`.

Here's a suggestion on how to handle it:

```python
sensorCount = rscpFindTagIndex(info, "BAT_DCB_NR_SENSOR")
if sensorCount is None:
    sensorCount = 0  # Or handle this case in some other appropriate way
for sensor in range(0, sensorCount):
    ...
```

This will prevent the error by setting `sensorCount` to 0 if it's `None`. Depending on your use case, you might want to log a warning or take some other action if `sensorCount` is `None`.

Would you like to apply this fix or investigate further?

eikowagenknecht, 2023-09-26:
Same problem here with a S10 X (see also https://github.com/torbennehmer/hacs-e3dc/issues/45).

closed 2023-10-14 by vchrisb