fsantini/python-e3dc #72: Home Assistant integration

kbabioch, 2023-04-11:
Just wondering: Is anyone working in a Home Assistant integration based upon this library already? Seems like the best option to get the data into Home Assistant is RSCP,.in particular the energy related data.

chrisghu, 2023-04-15:
I have roughly integrated this fantastic lib into an custom component for home assistant to read certain tags via RSCP and process the signals within home assistant. The component is at the moment highly adapted to the installation of a S10 Pro with only a bunch of signals (power of grid, wall box, battery, pv and pv external, SOC,... ). The component is reading the signals from E3DC S10 every 30s via this lib. The only drawback is that the E3DC has to be in the same subnet as the home assistant instance, otherwise RSCP will not work for reading the values locally. If someone is interested in the custom component I can spend some time to clean up the code and to resolve some hard coded things. The config flow of the component is also not working at the moment.

svlsbrg, 2023-04-16:
Hello Chisgru, 

I am definitely also very interested in a custom component that uses pye3dc as a requirement. I would be very happy if you could provide your implementation. I would also be happy to offer myself as a tester. Especially interesting for me is the possibility to use pye3dc to set the charging and discharging power of the battery as a service. 

Greetings and thanks

Rene

torbennehmer, 2023-04-28:
I am currently working on an native RSCP integration for HA on my local S10, testers will be welcome as soon as I've done a few basic tests at home. I'll get you more updates over the weekends.

svlsbrg, 2023-04-28:
I am very happy to support.   Do you want to publish the custom component. Then I can test.

torbennehmer, 2023-04-30:
@svlsbrg I'll let you know as soon as it is test-ready. Right now I still have to fix a number of linter errors and similar warnings to get it past the commit validation. Once this is done, I'll let you know where to find the code.

torbennehmer, 2023-05-07:
@svlsbrg I have currently pushed two branches with my experimental version. If you already want to take a peek, you need to install them manually. I'm currently still running it inside a devcontainer, but you should be able to manually push the component to an existing installation by copying it into the custom components. 

1st, the HA branch can be found here: https://github.com/torbennehmer/ha-core/tree/e3dc 
The component is under https://github.com/torbennehmer/ha-core/tree/e3dc/homeassistant/components/e3dc

2nd, if you have an S10, you might want to use my e3dc python branch (consider it unstable though), otherwise the S10s won't be detected correctly. The service and the switches included are also built against this version, so if you want to test them, please use my experimental branch
It can be found here: https://github.com/torbennehmer/python-e3dc/tree/s10-fixes
RIght now, I'm discussing these changes with the lib owner, they are not yet cleaned up therefore.

If you work directly with my repo, please be aware, that I'm currently force-pushing the branches from time to time, just that you know.

svlsbrg, 2023-05-07:
Hello, thank you very much. I copied the integration to the custom components folder. Then I also copied your version of the library into the folder. Still, after restarting Home Assistant, the integration is not displayed. I don't know exactly what I did wrong. There is nothing in the log either. Other custom components worked like this.

BR
Rene

torbennehmer, 2023-05-08:
I just tried it myself, the error is simple, non-core components do need a version key in their mainfest file, also a translation library is missing. Look at the following share, replace `manifest.json` and put `en.json` to `translations/en.json` (you need to create the directory). That should get you going.

https://owncloud.nehmer.net/index.php/s/pWwfQBsgQ7Piod2

Configuration-wise, please keep in mind that only [local connections](https://github.com/fsantini/python-e3dc#local-connection) are supported at this time. Pure Web Connections are unsupported.

svlsbrg, 2023-05-08:
Works absolutely perfect until now. My predictive battery power limitation is now up and running. 

I will feedback if I discover any faults. 

For charge and discharge limitation I will try to setup number entities. 

Best regards 
Rene

torbennehmer, 2023-05-09:
Great to hear!

regarding the number entities: I have avoided to do so and kept only the service because the settings are genereally related to each other. This is only one call going out to E3DC, so far as I can tell at least. This means, if you have number entities, you probably have to take care to update them in tandem

torbennehmer, 2023-05-16:
While I'm working to get this integrated into HA Core (which will take some time): I have created a HACS version of the integration, which for now will lead development. It will register itself as "e3dcgit", so that I can keep it separate from any future core integration. You can find it at https://github.com/torbennehmer/hacs-e3dc.

You'll have to add the repo manually to HACS for the moment, before I can push it into HACS' default repository list, I first have to upload a few brand images (for which I again have to wait for a merge).

Use the address `https://github.com/torbennehmer/hacs-e3dc.git` as a manual repository link. If you have trouble, please contact me via the hacs-e3dc repo.

Supported right now:

- A lot of sensors around basic operations, mostly what you'll see in the cloud frontend of the unit. No historical data yet, no sepcial stuff like SoH of the batteries etc.
- If you want to use it with an S10, you'll have to use current Git of pye3dc until the next release (Issue #73, fixes in #75 and probably #74)
- A bunch of diagnostic sensors to get an idea on basic operations
- Web based config, only local connections work, internet connectivity is required for the S10 to authenticate you for the local connection.
- Switches for power saving and weather regulation
- Service to control maximum battery charge and discharge amounts

This is about the featureset of the initial release, more is to come as soon as I have all my bookkeeping for HACS in place. Please feel free to test it.

**Important:** Please file all problems related to my integration at https://github.com/torbennehmer/hacs-e3dc - the marvelous developers behind pye3dc are not responsible for what I'm doing there :-)

**Much more important:** Thanks and Kudos for the pye3dc developers for their ground work, without it, I would not have been able ramp up a basic HA integration in two weeks worth of evenings!

@vchrisb Would it be an idea to close this thread and link all projects mentioned in this thread (not only mine) in your README?

closed 2023-10-20 by vchrisb