fsantini/python-e3dc #88: Issue accessing web connecting

marius-klemm, 2023-10-01:
Dear community around python-e3dc,

first of all a big thank you for your efforts to make the e3dc data accessible via script. I have a S10-E3DC system at home myself and would like to query the data regularly. This is how I became interested in your project.

Since the data retrieval should also happen outside the network where the E3DC system is located, I am mainly interested in the web access.

Unfortunately, I encounter the following error message during testing:
`Traceback (most recent call last):
  File "C:\Users\Marius\PycharmProjects\pythonProject\python_e3dc_master\e3dc\_e3dc.py", line 383, in sendRequest
    self.rscp.connect()
  File "C:\Users\Marius\PycharmProjects\pythonProject\python_e3dc_master\e3dc\_e3dc_rscp_web.py", line 278, in connect
    if not self.isConnected(): raise RequestTimeoutError
python_e3dc_master.e3dc._e3dc_rscp_web.RequestTimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Marius\PycharmProjects\pythonProject\e3dc-connection.py", line 1, in <module>
    from e3dc import E3DC
  File "C:\Users\Marius\PycharmProjects\pythonProject\e3dc.py", line 9, in <module>
    e3dc = E3DC(E3DC.CONNECT_WEB, username=USERNAME, password=PASS, serialNumber = SERIALNUMBER, isPasswordMd5=False)
  File "C:\Users\Marius\PycharmProjects\pythonProject\python_e3dc_master\e3dc\_e3dc.py", line 112, in __init__
    self.get_system_info_static(keepAlive=True)
  File "C:\Users\Marius\PycharmProjects\pythonProject\python_e3dc_master\e3dc\_e3dc.py", line 642, in get_system_info_static
    self.deratePercent  = round(self.sendRequest( ('EMS_REQ_DERATE_AT_PERCENT_VALUE', 'None', None), keepAlive = True  )[2] * 100)
  File "C:\Users\Marius\PycharmProjects\pythonProject\python_e3dc_master\e3dc\_e3dc.py", line 393, in sendRequest
    raise SendError("Max retries reached")
python_e3dc_master.e3dc._e3dc.SendError: Max retries reached

Process finished with exit code 1`

I have inserted username, password into the script properly. The serial number of my device is "722136016186". I have integrated this as a string as well.

Do you have an idea where the problem is and what I can do to make the script work and get a response? 

Thanks a lot
Marius

PS: I hope it's okay and correct that I opened an issue for this. That seemed to make the most sense to me.

closed 2023-10-01 by marius-klemm

fsantini, 2023-10-02:
Did the issue fix itself?

marius-klemm, 2023-10-02:
Hi @fsantini,

yes, the issue is solved. The problem was sitting in front of the computer. Stupid mistake on my part. I had a file called e3dc.py in my project which was called instead of the library.

By accident I noticed this at some point and changed it - and now everything works fine for now.

Sorry for the disruption and thanks for this great library :-)

fsantini, 2023-10-02:
Glad to hear. It happened to me too once :)