fsantini/python-e3dc #102: Cell voltage data is empty

martincornejo, 2023-10-31:
By polling the battery data, the voltage of each cell is requested:

https://github.com/fsantini/python-e3dc/blob/75d3d41f80c4284cfd78d99836850dda73074cd6/e3dc/_e3dc.py#L1275-L1284

When processing the request, for some reason, `BAT_DCB_NR_SERIES_CELL` is 0, skipping the loop and returning an empty list. Iterating voltage data instead would return the right values.

vchrisb, 2023-11-04:
what e3dc system do you have?
the last change was made by @eikowagenknecht 
For my S10 Mini it still does work.

eikowagenknecht, 2023-11-06:
@vchrisb Yes, I changed that file. But the logic was the same before. I do not see how the change could have introduces this bug / inconsistency, especially if it still works on your system.

Old:

```py
            voltages_raw = rscpFindTagIndex(
                rscpFindTag(req, "BAT_DCB_ALL_CELL_VOLTAGES"), "BAT_DATA"
            )
            voltages = []
            seriesCellCount = rscpFindTagIndex(info, "BAT_DCB_NR_SERIES_CELL")
            for cell in range(0, seriesCellCount):
                voltages.append(round(voltages_raw[cell][2], 2))
```

New:

```py
            voltages_raw = rscpFindTag(req, RscpTag.BAT_DCB_ALL_CELL_VOLTAGES)
            if (
                voltages_raw is not None
                and len(voltages_raw) == 3
                and voltages_raw[1] != "Error"
            ):
                voltages_data = rscpFindTagIndex(voltages_raw, RscpTag.BAT_DATA)
                seriesCellCount = rscpFindTagIndex(info, RscpTag.BAT_DCB_NR_SERIES_CELL)
                for cell in range(0, seriesCellCount):
                    voltages.append(voltages_data[cell][2])
```

The range code itself has not been changed (besides using the enum instead of the string).

The enum value also is the same:
`    0x03800300: "BAT_DCB_NR_SERIES_CELL",` vs `    BAT_DCB_NR_SERIES_CELL = 0x03800300`

vchrisb, 2023-11-06:
Thank you @eikowagenknecht for looking into it.
Let's see which System @martincornejo is using.

martincornejo, 2023-11-06:
It's a S10 E Pro

vchrisb, 2023-11-13:
For my system I do get for  `BAT_DCB_NR_SERIES_CELL`  `13`, but the `voltages_data` does have `14` items, while the last item's value is `0`.

martincornejo, 2023-11-15:
This might be a problem specific to the S10 E Pro. A solution could be to include an `if` statement that checks if `BAT_DCB_NR_SERIES_CELL==0`. In that case, it should return all `voltages_data`. Not the nicest solution, but I can't think of something better right now

closed 2024-10-13 by fsantini