fsantini/python-e3dc #110: set_switch_onoff returns False all the time but on-action is executed, off-action never executed

ako-kft, 2024-03-11:
Dear all:
I appears the method set_switch_onoff always returns false, no matter what the arguments. However, if passed the correct actuator id and the "on" command it will switch on the actuator. Turning the actuator off appears not possible, no matter what arguments are passed.

Can anyone confirm this?
Thank you, Adrian

vchrisb, 2024-03-21:
Hi,
yeah https://github.com/fsantini/python-e3dc/blob/master/e3dc/_e3dc.py#L396 always evaluates to `on`.
Would you mind to try out replacing `cmd` https://github.com/fsantini/python-e3dc/blob/master/e3dc/_e3dc.py#L404 with `value` and try it out?
This way either `on` or `off` is send directly. 
I can't, as I don't have switches.

ako-kft, 2024-03-21:
Hi –

Perfect. Works like a charm. Thank you very much. Will you fix it in the library?

Best regards,
Adrian

vchrisb, 2024-03-21:
yes

closed 2024-03-21 by vchrisb