fsantini/python-e3dc #115: e3dc._e3dc_rscp_local.CommunicationError: UNKNOWN

torbennehmer, 2024-03-22:
Hi @fsantini,

one of my HomeAssistant Users is receiving `e3dc._e3dc_rscp_local.CommunicationError: UNKNOWN` errors and I've no idea how to further narrow it down. You can find the complete logs and stackdumps from the error at torbennehmer/hacs-e3dc#124. There has been a similar issue in the past at torbennehmer/hacs-e3dc#75 as well.

Both cases have in common that they are dealing with time data, the first one was requesting the db data for the current day, the second one the unit's UTC time (for timezone detection/validation). But that could be a complete coincidence.

Have you seen similar errors in the past? Would there be a way of sort of an extended debugging mode (whatsoever) which could dump what you're seeing on the protocol level? My problem is, that a number of HA users are not tech-savvy, getting them to do diagnostics on the pye3dc level is hard. So I'd be grateful for any idea here.

Torben

Cc: @Phoenix-DH

Phoenix-DH, 2024-03-22:
@torbennehmer Thanks

torbennehmer, 2024-03-22:
The problem seems to appear in a third case, again when querying historic data at torbennehmer/hacs-e3dc#113, at the bottom of the logs there is the same error:

```
2024-02-28 00:00:13.087 ERROR (SyncWorker_2) [custom_components.e3dc_rscp.e3dc_proxy] Communication error with E3DC: Max retries reached
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc.py", line 226, in sendRequest
    result = self.rscp.sendRequest(request)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc_rscp_local.py", line 117, in sendRequest
    raise CommunicationError(receive[2])
e3dc._e3dc_rscp_local.CommunicationError: UNKNOWN

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/e3dc_rscp/e3dc_proxy.py", line 29, in wrapper_handle_e3dc_ex
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/e3dc_rscp/e3dc_proxy.py", line 116, in get_db_data
    return self.e3dc.get_db_data_timestamp(timestamp, timespan_seconds, True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc.py", line 710, in get_db_data_timestamp
    response = self.sendRequest(
               ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/e3dc/_e3dc.py", line 237, in sendRequest
    raise SendError("Max retries reached")
e3dc._e3dc.SendError: Max retries reached
```

I'm running a simple

```python
e3dc.get_db_data_timestamp(timestamp, timespan_seconds, True)
```

here, where timespan is counting from midnight. Could this be the root cause? The call being too close to midnight thus not giving us data? Have you had such effects?

Phoenix-DH, 2024-03-22:
Will provide additional data in the other thread.