fsantini/python-e3dc #121: WebSocket Connection not Working again (wss://s10.e3dc.com/ws)

Jonsch1661, 2024-08-12:
I saw in recent issues that the websocket adress was changed because the connection was no longer working.

I am also timing out on connection attempts using both websockets
wss://s10.e3dc.com/ws and the old one wss://s10.e3dc.com/ws/

Any more changes to the websocket that could cause this or what am i missing?
My passed credentials and serial number are correct. i tested it with the workaround script posted in a previous issue.

cosenagon, 2024-08-13:
I have the exact same issue. Is there any way to fix this?

fsantini, 2024-08-14:
Uhm, the address in the javascript sources is still that, so I don't think that this is the problem. We'll have to investigate it more.

RenoFischa, 2024-08-18:
try setting isPasswordMd5=True
when you specify the password as cleartext

`e3dc_obj = E3DC(E3DC.CONNECT_WEB, username=username, password=password, serialNumber = serialNumber, isPasswordMd5=True)
`